pimoroni/inky #183: Incompatible with Raspberry Pi 5B?

Enverex, 2023-12-08:
Trying to run this with all the required packages throws this for any of the test scripts:

```
Requires the seaborn library:
    sudo apt install python3-seaborn

Or from pypi (slower, requires building):
    sudo python3 -m pip install seaborn
    You may also need to: sudo apt install libatlas-base-dev

Detected 7-Colour 800x480 (AC073TC1A)
Traceback (most recent call last):
  File "/home/enverex/git/inky/examples/7color/graph.py", line 72, in <module>
    inky.show()
  File "/usr/local/lib/python3.11/dist-packages/inky/inky_ac073tc1a.py", line 352, in show
    self._update(buf.astype('uint8').tolist())
  File "/usr/local/lib/python3.11/dist-packages/inky/inky_ac073tc1a.py", line 298, in _update
    self.setup()
  File "/usr/local/lib/python3.11/dist-packages/inky/inky_ac073tc1a.py", line 202, in setup
    self._gpio.setup(self.cs_pin, self._gpio.OUT)
RuntimeError: Cannot determine SOC peripheral base address
```

Despite the error, seaborn is installed - `python3-seaborn is already the newest version (0.12.2-1)`

carlwalton, 2024-01-25:
Can you give an estimation of when this work is likely going to be completed?

Gadgetoid, 2024-02-06:
Sorry, got stuck down by COVID and haven't been able to progress here. Unfortunately I can't venture any kind of timescales because they would be a lie. I'm dealing with too many unknowns.

Which display are you trying to use? Some (but not all) are working, and it would be useful to get some additional input to see if what works for me works elsewhere.

Currently Inky Impressions 7.3" works, but installing is a little but of a faff:

```
git clone https://github.com/pimoroni/inky -b gpiod
cd inky
./install.sh --unstable
```

This should also add some new lines to `/boot/firmware/config.txt` so that SPI gets out of the way of the chip-select pin, which the library wants to claim-

```
dtoverlay=spi0-0cs
dtoverlay=i2c1
dtoverlay=i2c1-pi5
```

carlwalton, 2024-02-06:
@Gadgetoid Sorry to hear about your bout with COVID - hope you're feeling better. A while after commenting here I read your post on the Pimoroni board and felt a little sorry for the mess of work Bookworm has given you. Great news though, this works on my Impression4! I do receive similar feedback as you stated in #182. Here's the output from image.py - I haven't attempted to modify anything yet. Image looks great though. Thanks for the work you're doing here!

```Detected 7-Colour 640x400 (UC8159)
/home/carl/.virtualenvs/pimoroni/lib/python3.11/site-packages/inky/inky_uc8159.py:242: UserWarning: SPI: Cannot disable chip-select!
  warnings.warn("SPI: Cannot disable chip-select!")
/home/carl/.virtualenvs/pimoroni/lib/python3.11/site-packages/inky/inky_uc8159.py:337: UserWarning: Busy Wait: Held high. Waiting for 1.00s
  warnings.warn(f"Busy Wait: Held high. Waiting for {timeout:0.2f}s")
/home/carl/.virtualenvs/pimoroni/lib/python3.11/site-packages/inky/inky_uc8159.py:337: UserWarning: Busy Wait: Held high. Waiting for 0.20s
  warnings.warn(f"Busy Wait: Held high. Waiting for {timeout:0.2f}s")
/home/carl/.virtualenvs/pimoroni/lib/python3.11/site-packages/inky/inky_uc8159.py:337: UserWarning: Busy Wait: Held high. Waiting for 32.00s
  warnings.warn(f"Busy Wait: Held high. Waiting for {timeout:0.2f}s")```

Gadgetoid, 2024-02-07:
Thank you! Feedback is very useful- one of the hardest things (apart from basically all my knowledge of the Pi ecosystem and GPIO being consigned to the bin) is not having enough people to bounce these changes off, and verify I'm not doing anything wildly silly.

Thank you also for your understanding- I hate having to appeal to the wider complexity of the problem, since it always feels like there was some way we (or I, specifically) could have been more proactive (or even made different choices about the migration) and avoided some of this mess. I could certainly ramble justifications for hours, but that unfortunately wont fix anything 😭 

(I was fortunate to only have pretty mild COVID symptoms, but it hit the whole family so it's been a rollercoaster)

BryanH, 2024-04-25:
The issue is the PI5 uses a the RP1 southbridge chip, making the old library obsolete. Fortunately, there is a drop-in replacement.

Here is the simple workaround:

```
sudo apt remove python3-rpi.gpio
pip3 install rpi-lgpi
```

Gadgetoid, 2024-04-26:
For better or worse it would be much more helpful if Pi 5 / Bookworm users tried https://github.com/pimoroni/inky/pull/182 and complained at me if/when it breaks.

kazzle101, 2024-05-26:
As a follow up to @BryanH comment
```
sudo apt remove python3-rpi.gpio
sudo apt install python3-rpi-lgpio
```
Also works on the Pi5.

helgibbons, 2024-09-11:
Closing this one as we've now merged the updated installer into main (but do let us know if anyone has any issues with the new installer).

There's updated install instructions in the readme: https://github.com/pimoroni/inky/blob/main/README.md

I've also updated our Inky pHAT and wHAT Learn Guides:
https://learn.pimoroni.com/article/getting-started-with-inky-phat
https://learn.pimoroni.com/article/getting-started-with-inky-what

closed 2024-09-11 by helgibbons