pimoroni/inky #187: It didn't blow up

johnstaveley, 2024-03-08:
When running this command:

curl https://get.pimoroni.com/inky | bash

You are told that the library should only be installed on PI5's. However in the item description:

https://shop.pimoroni.com/products/inky-impression-7-3?variant=40512683376723

There is nothing about the screen being restricted to raspberry pi 5s.

So either fix the item description or the item library.

Gadgetoid, 2024-03-08:
Where in the command output does it mention Pi 5?

johnstaveley, 2024-03-08:
It is written in red just before you have to type yes to confirm you want to go ahead

johnstaveley, 2024-03-08:
I tested this on a Pi4

johnstaveley, 2024-03-08:
Ah yes, sorry. I read that as 05, oops

closed 2024-03-08 by johnstaveley

Gadgetoid, 2024-03-08:
No worries, ironically the Pi 5 is where it *breaks* 😢