pimoroni/inky #190: Trouble using numpy from the installation on headless raspberry pi

theptrk, 2024-05-06:
Using a raspberry pi zero with a fresh install using Raspberry Pi Imager.
I then SSH'd into the raspberry pi.

Steps:
```
# create a virtual env
python -m venv env

# activate
source env/bin/activate

# download the library
pip3 install inky[rpi,example-depends]

# open python
python 

# try to import
> from inky.auto import auto

Traceback (most recent call last):
  File "/home/patricktran/env/lib/python3.11/site-packages/numpy/core/__init__.py", line 24, in <module>
    from . import multiarray
  File "/home/patricktran/env/lib/python3.11/site-packages/numpy/core/multiarray.py", line 10, in <module>
    from . import overrides
  File "/home/patricktran/env/lib/python3.11/site-packages/numpy/core/overrides.py", line 8, in <module>
    from numpy.core._multiarray_umath import (
ImportError: libopenblas.so.0: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/patricktran/env/lib/python3.11/site-packages/numpy/__init__.py", line 130, in <module>
    from numpy.__config__ import show as show_config
  File "/home/patricktran/env/lib/python3.11/site-packages/numpy/__config__.py", line 4, in <module>
    from numpy.core._multiarray_umath import (
  File "/home/patricktran/env/lib/python3.11/site-packages/numpy/core/__init__.py", line 50, in <module>
    raise ImportError(msg)
ImportError:

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.11 from "/home/patricktran/env/bin/python"
  * The NumPy version is: "1.26.4"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libopenblas.so.0: cannot open shared object file: No such file or directory

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/patricktran/env/lib/python3.11/site-packages/inky/inky.py", line 8, in <module>
    import numpy
  File "/home/patricktran/env/lib/python3.11/site-packages/numpy/__init__.py", line 135, in <module>
    raise ImportError(msg) from e
ImportError: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/patricktran/env/lib/python3.11/site-packages/inky/__init__.py", line 3, in <module>
    from . import inky                            # noqa: F401
    ^^^^^^^^^^^^^^^^^^
  File "/home/patricktran/env/lib/python3.11/site-packages/inky/inky.py", line 10, in <module>
    raise ImportError('This library requires the numpy module\nInstall with: sudo apt install python-numpy')
ImportError: This library requires the numpy module
Install with: sudo apt install python-numpy
```

[image]

helgibbons, 2024-05-07:
Have you tried the other suggestion - removing the `pip` version of numpy and installing the system version from `apt`?

```
pip3 uninstall numpy  # remove previously installed version
apt install python3-numpy
```

theptrk, 2024-05-07:
Oh interesting, i tried `apt install python-numpy` but got this error 
```
(env) patricktran@myraspberrypi:~ $ sudo apt install python-numpy
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python-numpy is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python-numpy' has no installation candidate
```

Then I noticed you mentioned `python3-numpy`. I installed that ok.

But if I try to import the package, I still receive an error relating to "python-numpy"
```
(env) patricktran@myraspberrypi:~ $ python
^[[APython 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from inky.auto import auto
Traceback (most recent call last):
  File "/home/patricktran/env/lib/python3.11/site-packages/inky/inky.py", line 8, in <module>
    import numpy
ModuleNotFoundError: No module named 'numpy'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/patricktran/env/lib/python3.11/site-packages/inky/__init__.py", line 3, in <module>
    from . import inky                            # noqa: F401
    ^^^^^^^^^^^^^^^^^^
  File "/home/patricktran/env/lib/python3.11/site-packages/inky/inky.py", line 10, in <module>
    raise ImportError('This library requires the numpy module\nInstall with: sudo apt install python-numpy')
ImportError: This library requires the numpy module
Install with: sudo apt install python-numpy
```

RetroZelda, 2024-05-16:
i had this same problem and i had to apt install libopenblas-dev

Gadgetoid, 2024-05-16:
> But if I try to import the package, I still receive an error relating to `python-numpy`

By default virtual environments will ignore system installed packages.

You need to use the `--system-site-packages` switch.

Eg:

```
 python3 -m venv --system-site-packages ~/.virtualenvs/AAAAARRRGGHHHHHH
 ```

That said, I believe you can now re-install numpy into the virtual env and it will find the system libraries, since `python3-numpy` pulls them in as dependencies.

theptrk, 2024-06-14:
> By default virtual environments will ignore system installed packages.
>
> You need to use the `--system-site-packages` switch.
>
> Eg:
>
> ```
>  python3 -m venv --system-site-packages [...]

It worked! 

Now I get a different issue but I'll look into that separately. Thanks!

```
(env) patricktran@myraspberrypi:~ $ python
Python 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from inky.auto import auto
>>> display = auto()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/patricktran/env/lib/python3.11/site-packages/inky/auto.py", line 83, in auto
    raise RuntimeError("No EEPROM detected! You must manually initialise your Inky board.")
RuntimeError: No EEPROM detected! You must manually initialise your Inky board.
```

helgibbons, 2024-09-11:
Closing this one as we've now merged the new installer into main - this should attempt to set up a virtual environment for you if you don't have one.

There's updated install instructions in the readme: https://github.com/pimoroni/inky/blob/main/README.md

I've also updated our Inky pHAT and wHAT Learn Guides:
https://learn.pimoroni.com/article/getting-started-with-inky-phat
https://learn.pimoroni.com/article/getting-started-with-inky-what

closed 2024-09-11 by helgibbons