pimoroni/inky #191: buttons.py doesn't work: RuntimeError: Failed to add edge detection

ctrlcctrlv, 2024-05-27:
I wrote a C version here: https://gist.github.com/ctrlcctrlv/5d75f1d8e7bb1dd32183074d06e3bbac

But you probably want to fix this, I'm not sure how to:

```
fred@kinomi:~/Workspace/inky/examples/7color $ ./buttons.py
buttons.py - Detect which button has been pressed

This example should demonstrate how to:
 1. set up RPi.GPIO to read buttons,
 2. determine which button has been pressed

Press Ctrl+C to exit!

Attaching falling edge watcher to pin 5 with bouncetime 250
Traceback (most recent call last):
  File "/home/fred/Workspace/inky/examples/7color/./buttons.py", line 42, in <module>
    GPIO.add_event_detect(pin, GPIO.FALLING, handle_button, bouncetime=250)
RuntimeError: Failed to add edge detection
```

2br-2b, 2024-09-08:
I had the same problem on my Rpi 3. I attempted to follow the steps [here](https://stackoverflow.com/a/78241513/10442089), but doing this and rebooting didn't make a difference, nor did running this as sudo, as suggested [here](https://forums.raspberrypi.com/viewtopic.php?t=205327).

helgibbons, 2024-12-24:
We've just merged some updates to buttons.py that might help here - if you're still having problems getting errors with Pi OS Bookworm and the most recent version of the library please let us know.

https://github.com/pimoroni/inky/blob/main/examples/7color/buttons.py

closed 2024-12-24 by helgibbons