pimoroni/inky #192: buttons.py on Pi 5 with Bookworm is just a tad screwy :)

ttreker, 2024-06-14:
_All meant in good humor_

Having read the post about the joys of **gpiod**, **Pi 5**, **Bookworm** and being force-fed **Python virual environment** development I am guessing the single-commit port to these splendid techs of the **buttons.py** module was an untested first pass.

### Some Background
I am new to **Raspberry Pi** but not embedded Linux (some **BeagleBone Black** some years back). So I am ready to maybe hack through some of these **button.py** issues with perhaps some guidance from you if you would be so kind as to provide it.

I have a **Raspberry Pi 5** with **Bookworm** and the **Inky Impression 7.3"**.
I am a developer with probably the right mix of experience to tackle hacking out a workable **button.py** that I would be more than happy to provide in a pull request were I to succeed.

### Where you could provide some help:
I have been flummoxed over trying to figure out what GPIOs correspond to buttons on the display. I know the comments in the **button.py** module states that the GPIO pins in the **Python** list are the correct ones for the **Pi 5** but something doesn't add up there when I compare them to running _pinout_ and _gpioinfo_ in a terminal. I have found the following discrepancies (assuming **Broadcom** numbering:

- There is no GPIO29, GPIO31 and GPIO36 on the header (there is a GPIO18).
- From _gpioinfo_ it is not clear to me (my thick brain) what pins the I2c and SPI are are taking up.

This last point isn't directly related to the problem at hand I suspect but it would help me build a picture of Raspberry Pi 5 embedded system development.

**To Sum Up:**
My (possibly naive) view is that if you could provide me with what **Pi 5** GPIOs/header pins are assigned to the **Inky Impression 7.3"** I may be able to sort out the **button.py** example under **gpiod** and maybe even supply a device tree overlay for all of these items.

(Of course then again I might just end up pulling my hair out :) )

Gadgetoid, 2024-06-14:
It was a tested first pass and then I asked for the pin labels to be changed 😭

Then I decided that using labels was a bad idea…

“GPIO_” labels were physical IIRC (god only knows why) so if you look up the small numbers on https://pinout.xyz/ and translate them to the “GPIO” numbers it should get you the right pins.

You can also use “pinctrl” (Pi’s weird only-we-can-bypass-gpiod utility) to check the state of pins and set their value. Buttons usually connect a pin to ground so assuming GPIO29 is really GPIO5 “pinctrl 5 ip dh” (I’m doing this by memory from my phone so bear with me) *should* set that pin up to work with a button. Then run “pinctrl” to dunk the button state, run it again with the button held, tada!

Nuclear option would also be to claim *all* the pins and just loop through them in Python 🤣

IIRC gpioinfo has a “consumer” column, but I’m not totally sure the I2C and SPI drivers on Pi play nice and actually show up there. Regardless, trying to claim a used pin will usually result in a nice cryptic exception.

We have so many libraries that I did a first pass at Inky, it was all great (afaict!) and moved on to the next thing and the next and as I moved around the great circle of agony I found and reported issues. Changed my mind. Found better ways. And so on- all of that learning needs to be applied here.

Switching to libgpiod erased a great deal of my working knowledge about how to deal with GPIO, so I’m reset back ten years distant and doing a lot of on the fly learning.

Genuinely appreciate any effort you put into tackling this and any problems you find or suggestions you might have. It’s not fun tackling all this in isolation without someone to check I’m not meandering off into some bonkers over complicated distraction.

ttreker, 2024-06-14:
I am on Chicago time (what zone are you?) I have had some sleepless nights so this time is not a normal time for me to be replying. I just saw your reply above.

Thanks for this info! This looks very helpful! It gives me a lot with which to run.

It sounds like you've have done an amazing job. I scrolled through the Pimoroni repositories! There are a lot of them!

A little background:
I got involved in this helping a friend put together a custom device based on the GitHub repo tvldz/storybook for his grandchild. I have only been working on it for about a week. (Everything has worked pretty well so far on the display side. Given what you have gone through it seems pretty impressive! The Pi with your display has run well.) The upshot is that I got the Pi 5 to help my friend through this project (the Pi 5 was a gift from him for helping him out). So I haven't delved into the Pi until just this last week and when the project is done I am not anticipating continuing with any serious Pi projects (although you never know).

However, I am a low-level guy having started my software career in the 80's in embedded systems where I wrote  a lot of embedded assembly and C. (I left professional embedded systems behind in the early 90s.) I am also very comfortable with Linux. I looked into writing a kernel module for the BeagleBone Black about 8 or 9 years ago but that project fizzled out. So kernel experience is not in-depth by any stretch. I know how cryptic and complex kernel-based systems are so I have no doubt about the magnitude of the gpiod battle you have undertaken. Your idea of sorting out the gpiod complexities in abstracted libraries is a good one.

I mention all this to give some context of the degree of help you can expect. I am going to aim to sort out the buttons issue and in the process educate myself on the current state of GPIO development on the Pi 5 on Bookworm under Python (I hate not having a reasonable understanding of how things work). I don't like hacking things together so I'll do my best to come up with a clean, thoughtful buttons framework with perhaps supporting device tree overlays as an approach.

So I think I might be able to contribute a little something here.

Just don't count on me necessarily going far beyond that :)

ttreker, 2024-06-14:
I got the buttons.py sample code working. Thanks for your help on that. I am going to fork inky and do a pull request with the fixed version (Rasberry Pi 5 anyways.) I augmented your comments a bit to clear things up a bit regarding the pins and gpios.

closed 2024-12-24 by helgibbons