pimoroni/inky #195: Impression 5.7" light blue get displayed as green

Indiana8000, 2024-06-28:
Hello,

My Impression 5.7" display light blue as green, all other colores are very good.
I also tried the dither.py:
- fast method make a very large white area with some blue dots (realy uggly).
- slow method looks near identical as if no hitherdither is used.
- Saturation have no impact on the color change itself.

Any ideas what I can do to improve this?

This should be a gradiant from light blue to dark blue:
[image]

Gadgetoid, 2024-06-28:
There's isn't really a "light blue" in Inky's palette so aggressive dithering is required to get there.

Someone created a tool for experimenting with dithering (https://github.com/pimoroni/inky/issues/126), and there are other methods out there. But with the combination of a low resolution screen and a very meh (I wish they'd just give us CMYKW) colour palette (which will always have that vintage pixel patina) it's always a tradeoff.

Still I'd have expected better results- what's your source image, can you reproduce it here?

Indiana8000, 2024-06-28:
Since I use it to display AI art, this is not a big problem for me. Lately I've noticed that I see more greenish images than blue. So I start experimenting with all 7 colors.

I just searched for "blue gradiant" and used the first match: https://getwallpapers.com/wallpaper/full/6/e/e/101557.jpg
Would be interesting how it looks on your device.