pimoroni/inky #198: pip error with latest raspbian: inky install script not installing examples

javl, 2024-07-23:
With changes made to the latest Rasbian, the Inky example bash script installer no longer works, with pip throwing the new  "externally-managed-environment" error.

I've fixed it for myself by editing the Inky bash script (line 833), adding the "--break-system-packages"  to the pip install line.

From:
`if ! sudo -H $PIP3_BIN install --upgrade "$piplibname"; then`
to:
`if ! sudo -H $PIP3_BIN install --upgrade "$piplibname" --break-system-packages; then`

Info on this error plus some solutions can be found here:
https://pimylifeup.com/python-externally-managed-environment/

BreadMakesYouFull, 2024-07-25:
This might be covered by https://github.com/pimoroni/inky/pull/182
Specifically:
https://github.com/pimoroni/inky/pull/182/files#diff-043df5bdbf6639d7a77e1d44c5226fd7371e5259a1e4df3a0dd5d64c30dca44f

emilyboda, 2024-08-27:
Is there any update on this?

helgibbons, 2024-08-29:
Hi Emily - assuming you're using Pi OS Bookworm you might have more success with the updated installer (and installation instructions) in #182 ?

emilyboda, 2024-08-29:
I did try to install with those instructions and received errors for the examples. Is there any update on when the full non-beta version that works with the newest version of raspberry pi OS will be released? Or is there any information about which old version of raspberry Pi OS works with the existing instructions?

helgibbons, 2024-08-30:
> I did try to install with those instructions and received errors for the examples. 

Please can you let us know the specific errors you're getting with the updated installer over on #182 ? Details of what Pi/Inky combo you have might also be handy.

helgibbons, 2024-09-11:
Closing this one as we've now merged the new installer into main (but do let us know if anyone has any issues with the new installer).

There's updated install instructions in the readme: https://github.com/pimoroni/inky/blob/main/README.md

I've also updated our Inky pHAT and wHAT Learn Guides:
https://learn.pimoroni.com/article/getting-started-with-inky-phat
https://learn.pimoroni.com/article/getting-started-with-inky-what

closed 2024-09-11 by helgibbons