pimoroni/inky #201: No EEPROM detected on Inky Impressions 4"

PaulMatthieu, 2024-09-01:
I'm using the following product: https://shop.pimoroni.com/products/inky-impression-4?variant=39599238807635

And running inky v1.5.0 on python 3.11.2 on a raspberry pi 4

I get the following error:
```
>>> from inky import auto
>>> auto()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/turing/display/venv/lib/python3.11/site-packages/inky/auto.py", line 83, in auto
    raise RuntimeError("No EEPROM detected! You must manually initialise your Inky board.")
RuntimeError: No EEPROM detected! You must manually initialise your Inky board.
```

I'm _fairly sure_ the display is properly connected as it used to run fine before and I haven't touched it since. If there is any way to verify the connection (physically, it appears to be well seated on the GPIO pins) let me know.

Now doing the following works for me:
```
>>> from inky import Inky7Colour as Inky
>>> d = Inky()
```

But the size of the screen then appears to be wrong since there is a black bar on the display whenever I display something on it. I don't see many other options to initialize manually the display (there's `Inky_Impressions_7` but it seems to apply to the 7" model).

Would you please kindly let me know:
- Why isn't the display automatically detected anymore ? Or how can I further debug this EEPROM issue
- If I have to initialize it manually, how can I do so with an Impressions 4" so that the right size is used

Thanks

PaulMatthieu, 2024-09-01:
- If I have to initialize it manually, how can I do so with an Impressions 4" so that the right size is used

I don't find this to be very well documented but it appears the correct way to manually initialize an Inky Impression 4" is:

```
from inky import Inky7Colour as Inky
d = Inky(resolution=(640, 400))
```

still would love to know why the EEPROM stopped working

tajacks, 2024-10-28:
I also ran into this. Regarding your comment:

>'m fairly sure the display is properly connected as it used to run fine before and I haven't touched it since

Was it working sometime before this commit? 👇🏻 

https://github.com/pimoroni/inky/commit/23332f4a6f6d20f701dd7f6484b5a49bdbe14b6b#diff-432f16f0c031f58f5068d16e0ff94812465025ca1336153fb9c58fd751e09331R29

That's the last relevant commit I see related to the Impression 4" & EEPROM

admga, 2025-01-02:
Having the same issue on a Impression colour 5.7" with a Raspberry Pi 4.

RuntimeError: No EEPROM detected! You must manually initialise your Inky board.

admga, 2025-01-02:
Moved to a Raspberry Pi Zero 2W with Debian 11, but still getting the "No EEPROM" message. Is my board damaged? Last time I checked (a few months ago) it was working fine. Am I missing something?

admga, 2025-01-15:
> Have you tried manually enabling I2C and SPI in sudo raspi-config (or the
> graphical utility?) They're under 'Interfacing Options'.

Got it working. Thanks!