pimoroni/inky #202: "pins we need are in use" error when attempting to draw image to Inky Impressions 7.3"

FoxtrotPerry, 2024-09-12:
# Issue Description

I bumped my project to `inky` 2.0.0 and added `gpiod` and `gpiodevice` to the [pyproject.toml](https://github.com/FoxtrotPerry/sky-pi/blob/9741d0add6ca3122e8ffa81991ba1eeb6bc5496f/sky-pi-draw/pyproject.toml#L27) and encountered the following output:

```
Detected 7-Colour 800x480 (AC073TC1A)
Woah there, some pins we need are in use!
  ⚠️   Chip Select: (line 8, GPIO8) currently claimed by spi0 CS0
```

# Configuration Context

- This error was encountered running on a Raspberry Pi 3 running Raspberry Pi OS Bookworm (64-bit).
- This error did not occur when using `RPi.GPIO` version 0.7.1 and `inky` version `1.5.0`.

FoxtrotPerry, 2024-09-12:
Was able to find that the error is coming from [gpiodevice](https://github.com/pimoroni/gpiodevice-python/blob/d8e311580e08fc65227c7224ee095d42eb4c9886/gpiodevice/__init__.py#L47). I'm not familiar enough with raspberry pi architecture to know if is happening because I'm using a pi 3 instead of a 5. [Judging by the code](https://github.com/pimoroni/gpiodevice-python/blob/d8e311580e08fc65227c7224ee095d42eb4c9886/gpiodevice/__init__.py#L74) though, it seems like `gpiodevice` at least found a suitable gpiod chip on the pi 3.

Gadgetoid, 2024-09-12:
We probably need some better documentation around this, but since pins are now exclusive but Inky wants to control SPI chip select from software you might need to add the following to config.txt:

dtoverlay=spi0-0cs

crogers3, 2024-09-14:
I'm running into the same issue after upgrading to inky 2.0.0 on a Raspberry Pi Zero W. After setting the config option above, I'm encountering a different but similar error:

```
Chip Select: (line 8, SPI_CE0_N) currently claimed by spi-bcm2835
```

FoxtrotPerry, 2024-09-14:
I'm trying the suggested fix right now as well and will report back with whether it worked for me or not!

FoxtrotPerry, 2024-09-14:
Issue was resolved after adding `dtoverlay=spi0-0cs` to my `/boot/firmware/config.txt` as @Gadgetoid suggested! For those who might come across this issue in the future - my pi's `config.txt` was completely stock aside from a couple of unrelated HDMI entries.

closed 2024-09-14 by FoxtrotPerry

jmkerloch, 2024-11-11:
> I'm running into the same issue after upgrading to inky 2.0.0 on a Raspberry Pi Zero W. After setting the config option above, I'm encountering a different but similar error:
> 
> ```
> Chip Select: (line 8, SPI_CE0_N) currently claimed by spi-bcm2835
> ```

I still have this issue even after adding `dtoverlay=spi0-0cs` to `/boot/config.txt`

I also tried to create a file `/boot/firmware/config.txt`but without any luck.

Did anybody manage to fix this ?

jmkerloch, 2024-11-13:
> > I also tried to create a file `/boot/firmware/config.txt`but without any luck.
> 
> Does the file already exist? Does the `/boot/firmware` directory? 🤔 I can't remember when it was introduced!
> 
> spi0-0cs _should_ work for Pi Zero, though it's been a while since I fired one up 😬

In Bulleye it's seems that the file and directory doesn't exists.

I will try with a clean install on Bookworm and Bulleye.

jmkerloch, 2024-11-21:
> In Bulleye it's seems that the file and directory doesn't exists.
>
> I will try with a clean install on Bookworm and Bulleye.

I retry with a clean install on Bulleye and it worked. I think there might be a daemon service that could have use the port.

disruptivepatternmaterial, 2025-05-03:
I am seeing Chip Select 1: (line 16, GPIO16) currently claimed by lg on a fresh install of spextra6 13, any ideas how I fix this one?

jlplenio, 2025-09-04:
This was the fix for my old Pi Zero W: https://www.reddit.com/r/raspberry_pi/comments/1hhwu95/comment/m4qrzc6/