pimoroni/inky #205: Lots of "Busy Wait: Held high" warnings with every screen redraw (Inky Impression)

peepay, 2024-10-09:
Hello, I contacted support and they just confirmed the same behavior and asked me to report an issue here.

With the 5.7" Inky Impression and the latest version (2.0.0) of the Inky library, the commands output a lot of warnings with every screen redraw:

`(pimoroni) pi@raspberrypi:~/inky/examples/7color $ python3 image.py --file ~/photo.jpg`
`Detected 7-Colour (UC8159)`
`/home/pi/.virtualenvs/pimoroni/lib/python3.11/site-packages/inky/inky_uc8159.py:331: UserWarning: Busy Wait: Held high. Waiting for 1.00s`
`  warnings.warn(f"Busy Wait: Held high. Waiting for {timeout:0.2f}s")`
`/home/pi/.virtualenvs/pimoroni/lib/python3.11/site-packages/inky/inky_uc8159.py:331: UserWarning: Busy Wait: Held high. Waiting for 0.20s`
`  warnings.warn(f"Busy Wait: Held high. Waiting for {timeout:0.2f}s")`
`/home/pi/.virtualenvs/pimoroni/lib/python3.11/site-packages/inky/inky_uc8159.py:331: UserWarning: Busy Wait: Held high. Waiting for 32.00s`
`  warnings.warn(f"Busy Wait: Held high. Waiting for {timeout:0.2f}s")`
`(pimoroni) pi@raspberrypi:~/inky/examples/7color $ `

The screen redraws correctly, but the warnings do concern me.

Note: The warnings shown here are produced after running the image.py file from your examples, but the same warnings show even when I run the commands to load and display the image manually in python myself, so it is not limited only to the provided image.py file.

I checked the source at the referenced line and a related comment says "_If the busy_pin is \*high\* (pulled up by host) then assume we're not getting a signal from inky and wait the timeout period to be safe._" but I am not sure what that means.

grandmastermarco, 2024-11-13:
same for me. 
new setup on same hardware (RPi Zero 2W Bookworm 64bit headless - new os)
lots of error messages "...Busy Wait: Held high...." 
with example 7color scripts and also custom scripts.

but screen responds normally.

Gadgetoid, 2024-11-13:
> Interestingly, if there are more screen redraws in a single script (such as in the cycle.py file from your examples), only the first redraw produces the warnings.

This is definitely interesting.

Broadly the warnings are only there so we know this is happening and can use it as an extra diagnosis step if the display is *not* working. If it is working then they're safe to ignore.

I believe you can use Python's `warnings` module to suppress them:

```python
import warnings
warnings.filterwarnings("ignore")
```

kedare, 2024-11-22:
Confirmed on a new raspberry pi zero 2wh + inky impression73, on the example scripts:

```
python stripes.py -t impressions73
Failed to detect an Inky board. Trying --type/--colour arguments instead...

/home/kedare/.virtualenvs/pimoroni/lib/python3.11/site-packages/inky/inky_ac073tc1a.py:285: UserWarning: Busy Wait: Held high. Waiting for 1.00s
  warnings.warn("Busy Wait: Held high. Waiting for {:0.2f}s".format(timeout))
/home/kedare/.virtualenvs/pimoroni/lib/python3.11/site-packages/inky/inky_ac073tc1a.py:285: UserWarning: Busy Wait: Held high. Waiting for 0.40s
  warnings.warn("Busy Wait: Held high. Waiting for {:0.2f}s".format(timeout))
/home/kedare/.virtualenvs/pimoroni/lib/python3.11/site-packages/inky/inky_ac073tc1a.py:285: UserWarning: Busy Wait: Held high. Waiting for 45.00s
  warnings.warn("Busy Wait: Held high. Waiting for {:0.2f}s".format(timeout))
```

peepay, 2024-11-22:
> Failed to detect an Inky board.

That's a fundamental difference compared to my issue - the board is detected in my case.

Does it actually redraw the screen correctly for you, or not?

kedare, 2024-11-22:
Yes it does when I force via `-t`

peepay, 2024-11-25:
Okay, I did not have to force anything, though.

Let's leave it up to the Pimoroni staff to say whether it's related or not.

dnywh, 2025-04-01:
Same here with a brand new Inky Impression 4" and Raspberry Pi Model B+ v1.2. The display rendered the example files fine but would throw a lot of warnings.

Weirdly, the display stopped rendering the example files after a short while, maybe after 4–5 re-renders. Scripts that didn't rely on rendering (like buttons.py) continued to work fine.

I managed to get the screen to begin rendering again by increasing the timeout, as mentioned [here](https://forums.pimoroni.com/t/inky-impression-5-7-always-busy-wait/24029/7), but that doesn't fix the warnings issue. And seems very hacky to begin with.

kevinmcaleer, 2025-04-23:
I'm getting too on the HUGE 13.3" Spectra. Image draws fine after about 20 seconds

helgibbons, 2025-04-29:
I think the latest release should be less spammy with the warnings. Let us know if folks are still having issues!

closed 2025-05-20 by helgibbons

dnywh, 2025-06-12:
If it’s helpful, I’ve moved my 4" Inky over to a new Raspberry Pi W Zero 2, so got another chance to install from scratch.

In short: I’m still getting the warnings but it seems to be less spammy:

```bash
(pimoroni) pi@raspberrypi:~/inky/examples $ python3 name-badge.py --name Danny
Inky pHAT/wHAT: Hello... my name is:

Use Inky pHAT/wHAT as a personalised name badge!

Detected 7-Colour 640x400 (UC8159)
/home/pi/.virtualenvs/pimoroni/lib/python3.11/site-packages/inky/inky_uc8159.py:331: UserWarning: Busy Wait: Held high. Waiting for 1.00s
  warnings.warn(f"Busy Wait: Held high. Waiting for {timeout:0.2f}s")
/home/pi/.virtualenvs/pimoroni/lib/python3.11/site-packages/inky/inky_uc8159.py:331: UserWarning: Busy Wait: Held high. Waiting for 0.20s
  warnings.warn(f"Busy Wait: Held high. Waiting for {timeout:0.2f}s")
/home/pi/.virtualenvs/pimoroni/lib/python3.11/site-packages/inky/inky_uc8159.py:331: UserWarning: Busy Wait: Held high. Waiting for 32.00s
  warnings.warn(f"Busy Wait: Held high. Waiting for {timeout:0.2f}s")
```

So just three warnings these days. And the image draws fine.

This was testing using the GPIO header straight on a Pi, as well as via a GPIO breakout on a breadboard (with jumper cables).