pimoroni/inky #217: Can't find a driver this display when using display.auto()

jonbca, 2025-04-25:
I have a new Inky Impression 7.3" (2025 edition). Connected up to RPi 4B using the provided GPIO extender and receive the following error:

```
>>> from inky.auto import auto
>>> display = auto()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jabourbih/.virtualenvs/pimoroni/lib/python3.11/site-packages/inky/auto.py", line 85, in auto
    raise RuntimeError("Can't find a driver this display.")
RuntimeError: Can't find a driver this display.
>>>
```

I have tried plugging and unplugging the display from the board and still no success. Is this device supported by inky?

Gadgetoid, 2025-04-25:
The library release is a little delayed due to chasing a couple of bugs.

You can use the unstable/development instructions in the README to install the library from GitHub.

Or possibly, in your venv:

```
pip install --force-reinstall git+https://github.com/pimoroni/inky
```

jonbca, 2025-04-25:
Woohoo! It works 🎉 

Thanks!

closed 2025-04-29 by helgibbons