pimoroni/inky #227: IndexError - index 254 out of bounds for axis 0 with size 6

distant-storm, 2025-06-14:
I am running the spectra6 examples and expected to everything to just work out of the box.
I can run the stripes example with the default shipping image included with the examples, but it gives the below error.

Detected Spectra 6 7.3 800 x 480 (E673)
Traceback (most recent call last):
  File "/home/web/inky/examples/spectra6/image.py", line 31, in <module>
    inky.set_image(resizedimage, saturation=saturation)
  File "/home/web/.local/lib/python3.9/site-packages/inky/inky_e673.py", line 322, in set_image
    self.buf = remap[numpy.array(image, dtype=numpy.uint8).reshape((self.rows, self.cols))]
IndexError: index 254 is out of bounds for axis 0 with size 6

The comic.py also gives this error.
I normally think it is something I have done, but would expect these to work out of the box. Maybe not the comic one if images are some permutation not known, but at least the example shipping one.

I saw some spectra branches on the repo but nothing indicating that I should switch to them.
Any ideas?

Gadgetoid, 2025-06-16:
It looks like a palette-mode image is being used with colours outside of the supported range of Inky Frame.

I can create a dummy image that fails using the following code:

```python
from PIL import Image

img = Image.new("P", (800, 480))

for x in range(256):
    for y in range(256):
        img.putpixel((x, y), (x, 0, 0))

img.save("test.png")
```

I have a suspicion that https://github.com/pimoroni/inky/pull/221 will probably fix this.

distant-storm, 2025-06-16:
> It looks like a palette-mode image is being used with colours outside of the supported range of Inky Frame.
> 
> I can create a dummy image that fails using the following code:
> 
> from PIL import Image
> 
> img = Image.new("P", (800, 480))
> 
> for x in range(256):
>     for y in range(256):
>         img.putpixel((x, y), (x, 0, 0))
> 
> img.save("test.png")
> I have a suspicion that [#221](https://github.com/pimoroni/inky/pull/221) will probably fix this.

The thing that really confuses me is that it came out the box with the shipping image on and it looks great.
I ran the stripes example, then thought running image.py with image in /Pimoroni/inky/examples/spectra6/images/shipping-picture-inky7.jpg
would actually put it back to how it was, because it is in the spectra6 directory.
So if the colours are wrong, why is there a broken code/images in the spectra6 directory, how did that get merged in?

distant-storm, 2025-06-20:
> It's been noted that the image you posted with the wrong colours looks like the screen has stopped refreshing a few seconds into its full update cycle. How long did it take to display?

This actually helped ! It prompted me to try a different Pi, so I used a zero 2 but in the process had to flash another card. Followed the instructions and actually used OS Bookworm and it worked perfectly out of the box. 
Swapped back to the old pi with same OS and still worked. I think my other card was running a different variant of PI OS like lite.