pimoroni/inky #230: Support for new 13.3" 1600x1200 display (SKU: PIM774)

omiya-bonsai, 2025-07-01:
Hello Pimoroni team,

I have the new "Inky Impression - 13.3" (2025 Edition)" but I cannot get it to work.

**Hardware Details:**
- **Product Name:** Inky Impression - 13.3" (2025 Edition)
- **SKU:** PIM774
- **Resolution:** 1600 x 1200 pixels
- **Technology:** E Ink Spectra 6®

I have updated the `inky` library to the latest version from GitHub using `pip install --upgrade git+https://github.com/pimoroni/inky.git`.

When I try to initialize the display with the following Python code:
```python
from inky.inky import Inky
inky_display = Inky(resolution=(1600, 1200), colour="multi")

I get this error:
Resolution 1600x1200 not supported!

Could you please provide the correct way to initialize this new display?

Thank you!

Gadgetoid, 2025-07-01:
If you're using the drivers directly, you'll need `from inky.inky_el133uf1 import Inky`.

Or auto-detect from the EEPROM as per the Spectra 6 examples in this repo: https://github.com/pimoroni/inky/blob/main/examples/spectra6/image.py

```
from inky.auto import auto
inky = auto(ask_user=True, verbose=True)
```

omiya-bonsai, 2025-07-01:
Hi Gadgetoid,

Thank you so much for your reply and the advice!

Actually, I managed to solve the issue just before receiving your response.

To handle different Inky display models, I implemented a try-except block to dynamically import the correct class, like so:

```Python
try:
    # Attempt 13.3-inch class
    from inky import InkyEL133UF1 as InkyImpression
    INKY_CLASS = "InkyEL133UF1"
except ImportError:
    try:
        # Attempt Inky Impression class
        from inky import Inky_Impressions_7 as InkyImpression
        INKY_CLASS = "Inky_Impressions_7"
    except ImportError:
        try:
            # Attempt 7-colour class
            from inky import Inky7Colour as InkyImpression
            INKY_CLASS = "Inky7Colour"
        except ImportError:
            # Attempt standard Inky class
            from inky import inky
            InkyImpression = inky.Inky
            INKY_CLASS = "Inky"
```
This approach successfully initialized the display.

I really appreciate you taking the time to help.

I'll go ahead and close this issue now.